tock/tock #4204: PWM mux might be too restrictive

alexandruradovici, 2024-10-16:
While trying to get several servos to work, we noticed that the `MuxPwm` might be a little too restrictive. I am opening this issue first, as we might be wrong about this and we would like some feedback.

Currently the `MuxPwm` take a list of `PwmPinUser` devices. These devices keep internaly the actual PWM pin as a type that implements the `Pwm::Pin` trait. The way the mux works is that it allows one single client to have an *inflight* action.

Say we have the following actions using two PWM users, `pwm_a` and `pwm_b`, both of them of type `PwmPinUser` but using different PWM pins: `pwm_a` pin 0 and `pwm_b` pin 1.

## How it works now

- `pwm_a` calls `start`
- the mux has no inflight action, so it starts the PWM signal on pin 0 and sets `pwm_a` and inflight
- `pwm_b` calls `start`
- the mux queues the request as it has an inflight action already for `pwm_a`

## How it should work

- `pwm_a` calls `start`
- the mux has no inflight action, so it starts the PWM signal on pin 0 and sets `pwm_a` and inflight
- `pwm_b` calls `start`
- the mux verifies if `pwm_b` uses a different (*and in our uses case it does*):
  - if *yes*, the mux starts the PWM signal on pin 1
  - if *no*, the mux queues the request as it has an inflight action already for `pwm_a` which uses the same pin as `pwm_b`

Is our assessment of the current code correct?

bradjc, 2024-10-16:
I agree that if `pwm_a` and `pwm_b` can be used concurrently then Tock shouldn't force uses of those PWM outputs to be serialized.

I think the first consideration is that each underlying `PwmPin` should have its own mux if individual pins can be configured separately and concurrently. This would match any other hardware resource (e.g., it isn't "spi" that is muxed, it's a particular SPI bus that needs to be muxed, and separate SPI busses would have separate muxes).

But that might not be realistic for something as relatively simple as PWM pins. Without having thought about this extensively, I think we could do a single mux that tracks individual pins separately if we have some way of knowing which pins are the same. But, I think we would need to extend the HIL somehow to allow the underlying PWM driver implementation to express whether separate pins can in fact be separately controlled or not.

bradjc, 2024-10-18:
After some more thought and discussion, it seems like we don't want a virtualizer and instead want a wrapper which provides a `PwmPin`s. 

```rust
struct SinglePwmPin<P: Pwm> {
  pwm: &P,
  pin: &P::Pin,
}

impl SinglePwmPin {
  fn new(pin) -> Self;
}

impl PwmPin for SinglePwmPin {
  fn start(..) {
    self.pwm.start(self.pin, ...)
  }

  ...
}
```