tock/tock #4220: I/O level isolation

Arslan8, 2024-11-02:
Like the title suggests, is there a way to achieve I/O level isolation between two processes? For simplicity, let's say, i have these two contrived applications on BLE33Sense.
App1:
```
#include <libtock-sync/services/alarm.h>
#include <libtock/interface/led.h>

int main(void) {
  // Ask the kernel how many LEDs are on this board.
  int num_leds;
  int err = libtock_led_count(&num_leds);
  if (err < 0) return err;

  // Blink the LEDs in a binary count pattern and scale
  // to the number of LEDs on the board.
  for (int count = 0; ; count++) {
    for (int i = 0; i < num_leds; i++) {
        libtock_led_off(i);
    }

    // This delay uses an underlying alarm in the kernel.
    libtocksync_alarm_delay_ms(250);
  }
}
```

App2:
```
#include <libtock-sync/services/alarm.h>
#include <libtock/interface/led.h>

int main(void) {
  // Ask the kernel how many LEDs are on this board.
  int num_leds;
  int err = libtock_led_count(&num_leds);
  if (err < 0) return err;

  // Blink the LEDs in a binary count pattern and scale
  // to the number of LEDs on the board.
  for (int count = 0; ; count++) {
    for (int i = 0; i < num_leds; i++) {
        libtock_led_on(i);
    }

    printf("I am app2");

    // This delay uses an underlying alarm in the kernel.
    libtocksync_alarm_delay_ms(400);
  }
}
```
App1 and 2 can both easily access GPIO, because there is IO isolation. I am guessing this is expected behavior but there should be some IO level isolation since apps are not trusted. One possible design that comes to my mind is modify capsules to give permission based on a static ACL (maybe a manifest-like design). Is there a recommended way or pre-existing mechanism to achieve this?

alistair23, 2024-11-02:
Tock has syscall permissions, so you can restrict what syscalls apps have access to. I think that would help you

lschuermann, 2024-11-02:
Thanks for raising this concern @Arslan8. I think that this is something that Tock should provide (and does!), but is not currently demonstrated by any reference board definition in the upstream repository.

In particular, to expand on @alistair23's comment, system call filtering/permissions seem to be the right tool for the job here. For a guide that explores these mechanisms, you might want to look at the Security Key tutorial in the Tock book: https://book.tockos.org/course/usb-security-key/key-hotp-access#step-2-permitting-syscalls-for-only-credentialed-apps

Finally, it should be noted that you can instantiate Tock capsules multiple times (and expose them under a different driver number). This can help if you want to deny all access to a given driver _instance_ to some process and instead expose a _different_ instance of that same type of peripheral. I don't believe that we have a mechanism to map the same driver number to multiple instances depending on the process that invokes a system call, but that might be an interesting space to explore as well.

Hope this helps!

lschuermann, 2024-11-10:
@Arslan8 Closed without prejudice, please feel free to re-open in case you have more questions!

closed 2024-11-10 by lschuermann