tock/tock #4227: nrf52840 ieee802.15.4 tx syscall returns errorcode invalid

tyler-potyondy, 2024-11-08:
## Issue

@alevy found that the standard transmit ieee802.15.4 test failed for the nrf52840 with an error code invalid. Looking further into this, the root cause was the following check in `capsules/extra/src/ieee802154/driver.rs` (line 918):

```Rust
if cfg.len() != 11 {
  return None;
}
```

The `cfg` ReadableProcessSlice is provided from userspace and contains configuration information needed for forming the packet that is to be transmitted. During the libtock-c sync/async rewrite, it appears the allowed buffer value was mistakenly changed from 11 to 27, causing this bug.

## Needed Fix

Update libtock-c ieee802154 function.

tyler-potyondy, 2024-11-08:
https://github.com/tock/libtock-c/pull/474 resolves the invalid errorcode when performing the transmit syscall. However, I now get an errorcode OFF when attempting the tx syscall.

The standard 15.4 userspace driver  _assumes_ (as far as I can tell and based on comments in the `.finalize` method) that the radio will always remain on. The prior nrf52840 ieee802154 radio driver defaulted the radio to being turned on when initialized. https://github.com/tock/tock/pull/3993 altered this to now default to OFF. This means that we now need to turn on the radio standard 15.4 userspace driver's finalize method.

alevy, 2024-11-09:
> This means that we now need to turn on the radio standard 15.4 userspace driver's finalize method.

One way or another, something like this. It's probably _not_ good from a power consumption perspective to turn the radio on until it's actually needed though, right? Perhaps instead the driver should lazily turn it on when an operation is first requested? Or some maybe externalize to userspace to control?

tyler-potyondy, 2024-11-09:
Agreed on the power front. The design gets a bit more complicated here since the PHY control (e.g. Power on/off) are not exposed to the 15.4 standard driver (meaning that externalizing to userspace won't work). Options would be:

1. Naively turn on when initing radio for standard 15.4 driver (worst of solutions, but simplest)
2. Add logic to a lower layer in the 15.4 stack that turns on the radio upon the first tx (would need to look further into this).

alevy, 2024-11-09:
> The design gets a bit more complicated here since the PHY control (e.g. Power on/off) are not exposed to the 15.4 standard driver

Isn't that under our control to change?

But agree that (2) is maybe even better.

tyler-potyondy, 2024-11-09:
Yes, we _could_ change that but it would be a fairly heavy modification to the interface. The current 15.4 standard driver holds a reference to an object implementing the `Mac` trait (which is our Mac Mux). The Mac Mux then in turn holds the radio object that possess control to turn on/off. 

As is, there is an abstraction such that the 15.4 driver is operating at the Link Layer without access to directly modify PHY layer parameters. I think this is _good_ from a "I want to send and receive bits" perspective as it is simple. This was the reason we needed the phy driver so that we could directly expose that layer's control to userspace.

closed 2024-11-12 by alevy