tock/tock #4245: ARMv6m: Register clobbering in Generic ISR causes exception exit to return to process rather than Kernel

vrindisbacher, 2024-11-21:
I'm opening this issue following the discussion in #4242.

In the v6m `generic_isr` exceptions handled when running a process seems to return to that process rather than the kernel. On [this line](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L116), we load the special MSP return value into `r0`, but unfortunately we clobber it on the [next line](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L119). When we return from the ISR [here](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L155C8-L155C10) I believe the value of `lr` is `0xFFFFFFFD` which will cause a return to the process rather than the kernel. 

I think it suffices to move `ldr r0, 200f` to line `155` before we [return from the exception](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L155). This is redundant but follows a similar pattern to what's done in `v7m` where we load the MSP return value `0xFFFFFFF9` into the link register unconditionally.

alevy, 2024-11-27:
Hrm... I can't find any documentation that supports using these magic values (`200f` and `300f`) to  compare against `LR` or why setting `r0` would make any difference for which mode to switch to on `bx lr`.

@ppannuto why is this implementation so different from what's being done in `cortex-v7m`? According to the docs, the exception entry/return behavior is very very similar except for maybe fewer hardware stacked registers and fewer return options (due to lack of floating point): https://developer.arm.com/documentation/dui0497/a/the-cortex-m0-processor/exception-model/exception-entry-and-return?lang=en

alevy, 2024-11-27:
Oh! I misread `200f` and `300f` as hex values rather than labels. OK, this makes sense, except of course that as you say, this it's never loaded to `lr` so doesn't make a lick of difference.

alevy, 2024-11-27:
In this case, the better solution seems to be to move `ldr lr, 200f` instead of into `r0`. `lr` is never clobbered (and really really shouldn't be) and this wouldn't introduce an extraneous instruction in the path where we just came from the kernel.

PR #4259 (2024-12-17): ARMv6: fix setting the CONTROL register
### Pull Request Overview

This pull request fixes a fault in `generic_isr` and `svc_handler` by setting the `CONTROL` register accordingly. Without this fix, after an interrupt, applications continue to work in privileged mode.

This fixes #4245 and fixes #4246.

### Testing Strategy

This pull request was tested using a Raspberry Pi Pico by @JADarius.

### TODO or Help Wanted

Feedback from @ppannuto and others that are more experienced in ARM architecture.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are requi [... cut]
closed 2024-12-17 by alevy