tock/tock #4246: ARMv6m: Thread mode not set to privileged execution in certain ISRs

vrindisbacher, 2024-11-21:
I'm opening this issue following the discussion in #4242.

I believe that the following ISRs need to set the special `CONTROL` register to `0` before exception return, similar to [the `systick_handler`](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L189) on [line 193](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L193)

- [Generic ISR](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L89)
     - I think that this is canceled out by #4245., since we return to the process rather than the kernel when coming from a process. 
- [SVC Handler](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L226)      
     - My hunch here is that we should be setting thread mode to privileged execution in the case we are coming from a process and need to return to the kernel to handle a syscall. I think setting the `CONTROL` register to `0` [here]((https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-m0/src/lib.rs#L233)
) should do the trick. Similar to how the `v7m svc handler`  does [here](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-v7m/src/lib.rs#L121).
    - It might also be a good idea to ensure that we set thread mode to unprivileged execution when we go from kernel to process. It seems like the `v7m svc handler` does this [here](https://github.com/tock/tock/blob/a1966b8ddaa1ce819b80f2f7bea466eb76e5b46c/arch/cortex-v7m/src/lib.rs#L90).

alexandruradovici, 2024-11-28:
Looking into it.

alexandruradovici, 2024-11-28:
Looking at the code, it seems that:
- `generic_isr` has the bug, fixing it now
- `svc_handler` from` cortex_m0` has the bug, but is never used, we use the `svc_handler_m0p` function

PR #4259 (2024-12-17): ARMv6: fix setting the CONTROL register
### Pull Request Overview

This pull request fixes a fault in `generic_isr` and `svc_handler` by setting the `CONTROL` register accordingly. Without this fix, after an interrupt, applications continue to work in privileged mode.

This fixes #4245 and fixes #4246.

### Testing Strategy

This pull request was tested using a Raspberry Pi Pico by @JADarius.

### TODO or Help Wanted

Feedback from @ppannuto and others that are more experienced in ARM architecture.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are requi [... cut]
closed 2024-12-17 by alevy