tock/tock #4249: The `.attributes` section is causing an oversized `.bin` image

ZhekaS, 2024-11-22:
The default `tock_kernel_layout.ld` is defining the `.attributes` section, which is supposed to be placed at the *end* of the flash address space. It creates a huge zero-filled "hole" in raw binary generated by `objcopy`. Sure, this section can be stripped off as well using `objcopy`, but there is no such an option in the currently provided build process and also the metadata stored there will be lost, which might be or not a big deal.

ZhekaS, 2024-11-25:
> I'm not sure whether or not I would consider this a bug/issue. In particular, this also has the design feature that all of the flash region is set to a known value (even the 'unused' regions), as-opposed to being e.g. not overwriting whatever happened to be there before flashing the Tock kernel.
> 
> Grand scheme of things, the `.bin` is still a pretty small file — is there a specific problem the larger `.bin` is causing?
> 
> n.b., downstream users can, of course, define their own layout. e.g., many have static apps, and thus have a completely different path to creating the final image.

The resulting `bin` file is always the size of the defined flash, that can be pretty large relative to the actual "payload". That is, I am looking at my config with flash/bin size of 1MB vs the actual Tock OS size of ~90KB. 
There are couple of problems I see with this:

1.  Programming time - instead of flashing just a small portion of the flash, the whole region needs to be overwritten. Also with the erased value in flash being `FF`, writing all to zeros is just .. unnecessary.
2.  Number (1) becomes particularly important, when loading the image happens at boot time (that is, relocating from XN flash to executable memory).
3. Then Tock OS is only a part of a larger "package" - that is a BIOS containing many different firmwares that, where the storage space might become important.

Sure, as I stated it is not a critical deal-breaker, as this section seem to be "optional" and can be stripped relatively easily.

ZhekaS, 2024-11-25:
On the other hand, I guess the `.prog` size in the specific layout can be deliberately defined as small as desired, not necessarily as the size of the actual flash. So maybe it is a non-issue after all

ppannuto, 2024-11-25:
Yeah, I think my instinct is that the 'default behavior of upstream' make sense to be the slightly more paranoid, defense-in-depth, always set all of flash to a known value. If specific use cases downstream have different use scenarios, changing is as you noted, easy.

I'm going to close this, as it seems resolved from the latest comment. But feel free to continue commenting and we can re-open if I misjudged.

closed 2024-11-25 by ppannuto