tock/tock #4272: Call for Tock 2.2 Release Testing

lschuermann, 2024-12-18:
We just tagged a [release candidate for Tock 2.2](https://github.com/tock/tock/releases/tag/release-2.2-rc1), and it's ready for testing!

As discussed in https://github.com/tock/tock/issues/3197, we're changing our testing strategy starting with this release. Instead of trying to test all boards, we will primarily focus on [Tier 1 boards](https://github.com/tock/tock/tree/master/boards#tier-1), and specifically the nRF52840DK which we already test as part of our new "Treadmill" hardware CI system. These efforts should allow us to more frequently test & release Tock's "core" components, without placing as large of a burden on maintainers and board contributors.

That said, we don't want to discourage anyone from testing! We'll use this issue for tracking these efforts, which supersedes https://github.com/tock/tock/issues/3197. If you find yourself with some time and a board on your hands, feel free to copy the "test comment template" below and post your results in a comment to this issue.

**Assuming that we don't find any issues or receive requests to extend the testing period, I plan to tag a release some time around Dec 25th (in about a week).**

## Changes in Tock 2.2

Over the next week I will compile a more complete changelog. In the meantime, from the release notes:

> With almost two years since Tock 2.1.1, it's about time for a new Tock release! Compared to 2.1.1, this release will contain 3882 commits made in 836 PRs by 90 contributers (of which 48 are new contributors!). This release is the first to support stable Rust for out-of-tree and (a subset of) the in-tree boards, adds new supported boards, and contains many other fixes and improvements. Some notable bug fixes are in the Cortex-M arch crate to prevent userspace applications from executing in privileged mode, and disabling unused PMP regions on RISC-V. A full change list will be included with the final release notes.

## Misc Release Todo

- [ ] Update changelog (@lschuermann)
- [ ] Update crate version numbers (see https://github.com/tock/tock/issues/3197#issuecomment-2248183406)

## Test Boards

Tier 1:
- [ ] [Hail](boards/hail/README.md): no test comment yet
- [ ] [Imix](boards/imix/README.md): no test comment yet (@lschuermann)
- [ ] [Nordic nRF52840-DK](boards/nordic/nrf52840dk/README.md): no test comment yet (@lschuermann)
- [ ] [Nano 33 BLE](boards/nano33ble/README.md) no test comment yet
- [ ] [Nano 33 BLE Rev2](boards/nano33ble_rev2/README.md): no test comment yet
- [ ] [BBC Micro:bit v2](boards/microbit_v2/README.md): no test comment yet
- [ ] [Clue nRF52840](clue_nrf52840/README.md): no test comment yet

Others:
- [ ] _Your board could be here!_

## Test Comment Template

<details>
<summary>Template:</summary>

- Tests run on most Boards
  - [ ] `examples/sensors`
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  - [ ] `examples/blink`
  - [ ] `examples/rot_client` and `examples/rot_service`
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

- nRF specific
  - [ ]  `examples/ble_advertising`
  - [ ]  `examples/ble_passive_scanning`

- Imix-specific
  - [ ] `examples/tests/imix`
  - [ ] `examples/ble-uart`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port:
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
  -  [ ] **In-Kernel Tests**
      - [ ] `boards/imix/src/test/aes_ccm_test.rs`
      - [ ] `boards/imix/src/test/aes_test.rs`
      - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
      - [ ] `boards/imix/src/test/linear_log_test.rs`
      - [ ] `boards/imix/src/test/log_test.rs`
      - [ ] `boards/imix/src/test/rng_test.rs`
      - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
          - [ ] `start()`
          - [ ] `start_rx()`
          - [ ] `start_with_app()`
          - [ ] `start_dual_rx()`
      - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`
      - [ ] `boards/imix/src/test/multi_alarm_test.rs`

 - Microbit-specific Tests
   - [ ] `examples/accel-leds`
   - [ ] `examples/music`
   - [ ] `examples/tests/ble/ble_advertise`

- Miscellaneous
  - [ ] **Scheduler Tests**
    - [ ] `whileone` + `blink`
    - [ ] restart `whileone` using process console
    - [ ] stop and start `whileone`
    - [ ] `whileone` + `blink` + `hello_loop` using MLFQ scheduler
</details>

lschuermann, 2024-12-18:
### nRF52840-DK

Automated test results from: https://github.com/tock/tock-hardware-ci/actions/runs/12399190636

- Tests run on most Boards
  - [x] `examples/sensors`
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  - [x] `examples/blink`
  - [ ] `examples/rot_client` and `examples/rot_service`
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region` (both RAM & flash overrun work)
  - [x] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

- nRF specific
  - [ ]  `examples/ble_advertising`
  - [ ]  `examples/ble_passive_scanning`

- Miscellaneous
  - [ ] **Scheduler Tests**
    - [ ] `whileone` + `blink`
    - [ ] restart `whileone` using process console
    - [ ] stop and start `whileone`
    - [ ] `whileone` + `blink` + `hello_loop` using MLFQ scheduler

lschuermann, 2024-12-24:
### Imix

- Tests run on most Boards
  - [x] `examples/sensors`
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  - [ ] `examples/blink`
  - [ ] `examples/rot_client` and `examples/rot_service`
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

- Imix-specific
  - [ ] `examples/tests/imix`
  - [ ] `examples/ble-uart`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port:
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
  -  [ ] **In-Kernel Tests**
      - [ ] `boards/imix/src/test/aes_ccm_test.rs`
      - [ ] `boards/imix/src/test/aes_test.rs`
      - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
      - [ ] `boards/imix/src/test/linear_log_test.rs`
      - [ ] `boards/imix/src/test/log_test.rs`
      - [ ] `boards/imix/src/test/rng_test.rs`
      - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
          - [ ] `start()`
          - [ ] `start_rx()`
          - [ ] `start_with_app()`
          - [ ] `start_dual_rx()`
      - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`
      - [ ] `boards/imix/src/test/multi_alarm_test.rs`

- Miscellaneous
  - [ ] **Scheduler Tests**
    - [ ] `whileone` + `blink`
    - [ ] restart `whileone` using process console
    - [ ] stop and start `whileone`
    - [ ] `whileone` + `blink` + `hello_loop` using MLFQ scheduler

lschuermann, 2024-12-28:
Update: it seems like there are issues with the 15.4 subsystems or tests on the nRF52840DK (and potentially Imix). To quote @tyler-potyondy:
> The nRFs makes sense it isn’t working since UDP is sending through the standard 15.4 driver (which is currently not working).

@tock/core-wg Should we try and address these concerns before the release? The nRF52840DKs were just successfully used as part of the OpenThread tutorial, hence there is a good level of confidence that this part works. I don't have a good idea of how invasive the required changes to the remaining 15.4-related userspace and kernel subsystems are.

alistair23, 2024-12-29:
The Apollo3 boards are currently heavily tested, I think they should be ticked off under `Test Boards`. I'm not going to have a chance to run the example tests, but I'm testing a range of components at the moment, so I know applications, SPI, I2C, LED, alarms, GPIOs and flash all work

lschuermann, 2024-12-29:
@alistair23 This is great news! Can you maybe try and add a test comment with the subsystems that you know are working? I don't own one of those boards, and it'd be great to have a comment with slightly more information than "heavily tested" to point to, but in any case I'm happy to add this to the above!

lschuermann, 2025-01-03:
# LiteX Sim / Arty

This combines the test results from the LiteX Sim and Arty platforms, which use gatewares generated from the same LiteX revision (and hence should be representative of each other).

- Tests run on most Boards
  - [ ] `examples/sensors`
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  - [ ] `examples/blink`
  - [ ] `examples/rot_client` and `examples/rot_service`
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

- nRF specific
  - [ ]  `examples/ble_advertising`
  - [ ]  `examples/ble_passive_scanning`

- Imix-specific
  - [ ] `examples/tests/imix`
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
  - [ ] `examples/ble-uart`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port:
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
  -  [ ] **In-Kernel Tests**
      - [ ] `boards/imix/src/test/aes_ccm_test.rs`
      - [ ] `boards/imix/src/test/aes_test.rs`
      - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
      - [ ] `boards/imix/src/test/linear_log_test.rs`
      - [ ] `boards/imix/src/test/log_test.rs`
      - [ ] `boards/imix/src/test/rng_test.rs`
      - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
          - [ ] `start()`
          - [ ] `start_rx()`
          - [ ] `start_with_app()`
          - [ ] `start_dual_rx()`
      - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`
      - [ ] `boards/imix/src/test/multi_alarm_test.rs`

 - Microbit-specific Tests
   - [ ] `examples/accel-leds`
   - [ ] `examples/music`
   - [ ] `examples/tests/ble/ble_advertise`

- Miscellaneous
  - [ ] **Scheduler Tests**
    - [ ] `whileone` + `blink`
    - [ ] restart `whileone` using process console
    - [ ] stop and start `whileone`
    - [ ] `whileone` + `blink` + `hello_loop` using MLFQ scheduler

tyler-potyondy, 2025-01-04:
On the nrf52840, the following 802.15.4 tests are now passing (with #4291): 

- [x] `examples/tests/ieee802154/radio_tx`
- [x] `examples/tests/ieee802154/radio_rx`
- [x] `examples/tests/ieee802154/radio_tx_raw`

Additionally, OpenThread functionality (Tock MTD attaching to an OpenThread router) has been tested:

- [x] `examples/openthread/openthread_hello`

UDP transmit functionality is currently broken with a bug / inconsistency between the kernel and `libtock-c` implementation. When executing the transmit syscall, the `libtock-c` application fails to provide the src address and fails the error check that occurs for the transmit syscall.

alistair23, 2025-01-06:
> @alistair23 This is great news! Can you maybe try and add a test comment with the subsystems that you know are working? I don't own one of those boards, and it'd be great to have a comment with [...]

UART, LoRa, SPI, I2C, LED, alarms/timers, GPIOs, sensors, flash, K/V store, applications, large applications, application signature checking, asynchronous process loader and low power modes all work

lschuermann, 2025-01-06:
> UDP transmit functionality is currently broken with a bug / inconsistency between the kernel and `libtock-c` implementation. When executing the transmit syscall, the `libtock-c` application fails to provide the src address and fails the error check that occurs for the transmit syscall.

@tyler-potyondy Thanks! I added this to the release notes in #4280.

lschuermann, 2025-01-06:
We landed Tock 2.2! https://github.com/tock/tock/releases/tag/release-2.2 :tada:

closed 2025-01-06 by alevy