tock/tock #4302: Add a DebugWriter with a non-UART backend

ZhekaS, 2025-01-10:
Currently it looks like the `DebugWriter`/`DebugWriterWrapper` can only be implemented around a UART-like interfaces. In some cases it quite inconvenient. For example one might want to have `debug!()` prints to be dumped into a memory buffer, some semi-hosting mechanism or other simple output devices (my real-life example is a simulator with a single byte memory mapped write location which can be though as an "infinite" output FIFO). These mechanisms do not require `Receive` functionality at all, nor they require asynchronous capabilities (such as Transmit/ReceiveClient support and such).
The work-around, is to implement a Uart-like type for the simpler interfaces, mocking the non-existent functionality, but it looks like extra work for no extra benefit.

lschuermann, 2025-01-11:
I remember @kupiakos / @granaghan (and maybe others?) to talk about something similar a while back. One of the outcomes of that discussion was being able to serialize things such as process state and store that in flash for further debugging:

https://github.com/tock/tock/blob/c51466a7dfee238aa972cd8a3cb523e271d2b007/arch/rv32i/src/syscall.rs#L85-L107

Maybe they can expand on whether something similar is also possible for `DebugWriter`?

ppannuto, 2025-01-15:
While I agree the implementation details are a bit awkward, for something immediately usable, the SeggerRTT component is a good example of using an 'in-memory buffer' as a debug output.

bradjc, 2025-01-22:
It's generally easier to implement an asynchronous interface using a synchronous operation than vice-versa. Given Tock's design is asynchronous, we've found it to be a reasonable trade off to add a bit of overhead to cases where the underlying operation is synchronous.

Looking at the Segger RTT code (https://github.com/tock/tock/blob/master/chips/segger/src/rtt.rs), it's not too many lines of code to implement `uart::Transmit` above a sync operation. And it would be fairly straightforward I think to separate out the async-to-sync portion of the code into its own capsule that could be re-used.

ZhekaS, 2025-03-25:
Recently I also encountered an issue stemming from the fact the debug interface is asynchronous, where the actual message order is not exactly reflecting the correct order of the debugged events.
I have drafted a following change: https://github.com/tock/tock/pull/4382 
It is not very clean, but keeps the existing functionality in place, however allows implementing something like this:
```rust
pub struct SyncDebugWriter;

impl Write for SyncDebugWriter {
    fn write_str(&mut self, s: &str) -> core::fmt::Result {
        self.write(s.as_bytes());
        Ok(())
    }

    fn write_fmt(&mut self, args: core::fmt::Arguments<'_>) -> core::fmt::Result { 
        write(self, args)?;
        Ok(())
    }
}

impl DebugWriterTrait for SyncDebugWriter {
    fn publish_bytes(&self) -> usize {
        0
    }
}

impl IoWrite for SyncDebugWriter {
    fn write(&mut self, buf: &[u8]) -> usize {
        let out_reg = 0x4000 as *mut u8;
        for c in buf.iter() {
            unsafe { out_reg.write_volatile(*c) };
        }
        buf.len()
    }
}

```

And passing it to kernel:
```rust
    let debug_writer = static_init!(
        utils::SyncDebugWriter,
        utils::SyncDebugWriter
    );

    kernel::debug::set_debug_writer_wrapper(debug_writer);
```

Not sure if it is something to be converted to an actual PR in the current form

bradjc, 2025-03-25:
It would be helpful to understand what about the async interface leads to debug messages being out of order. There is nothing inherent about an async interface that means ordering cannot be preserved.

Is the issue the internal buffer in `DebugWriterWrapper` is filling up? I think adding a trait like in #4382 is reasonable, but I would like to better understand what the different requirements and use cases are for different `DebugWriter` implementations.

ZhekaS, 2025-03-25:
Sorry, I tried to come up with some qemu-based example, but couldn't figure it out. But I guess a flow in a quesion ight be something like this:
```rust
    debug!("Before event");
    led_on();   // <----immediately observable
    debug!("After event");
```
Since the debug writer in most cases is interrupt-driven, both prints might show up after the LED is turned on. One can argue, that it can happen in any case due to different speed of different peripherals, but in this case if we put a breakpoint (or a busy loop) right after the second `debug!`, we will hit it and see the LED is on but neither prints have shown up.

bradjc, 2025-04-09:
Ah ok, so out of order with respect to other operations in the system, not with the messages themselves out of order from the order of the `debug!()` calls. It makes sense that can happen.

Using an async UART-style interface should still be fine even for this reordering case, at least for one `debug!()` call. An effectively sync underlying writer mechanism can complete the write on the downcall (just as it would if the UART interface was sync). Once the write has completed, then the writer can set a deferred call. Then, the write call returns back to the `debug!()` call site with the invariant that the message has been sent at that point, ensuring the order. Once the deferred call has happened the callback is triggered, completing the full operation.

There are two issues with that, however:
1. With the UART interface, the buffer is held by the UART implementation until the deferred call runs. Any `debug!()` calls that happen before then will either be rejected or enqueued somehow until the deferred call runs (and the callback returns the buffer). The messages may be printed eventually, but will be out-of-order with respect to any events between the first `debug!()` call and control returning to the core kernel loop.
2. The current `DebugWriter` implementation that uses an internal ring buffer can insert multiple calls to the underlying UART implementation if the to-send message is longer than the buffer used to pass data to the UART implementation.

I'm convinced we need a debug writer that uses a fully synchronous interface for boards that have a synchronous mechanism to send debug messages.