tock/tock #4304: kernel/hil/uart: Add `EBUSY` error for `configure` in case of ongoing operation

lschuermann, 2025-01-13:
I think this is fine for now, but we should extend the UART HIL to be able to report `EBUSY` in this case. Silently interrupting an ongoing transmission for reconfiguration is not great.

_Originally posted by @lschuermann in https://github.com/tock/tock/pull/4300#discussion_r1913212895_

jrvanwhy, 2025-01-13:
The [UART HIL says](https://github.com/tock/tock/blob/699411a5539f82059b1709741d0c95bd59b37497/doc/reference/trd-uart.md#2-configuration-and-configure):

> Any configuration-change must not apply to operations started before this change. The UART implementation is free to accept a configuration change and apply it with the next operation, or refuse an otherwise valid configuration request because of an ongoing operation by returning `ErrorCode::BUSY`.

lschuermann, 2025-01-13:
Aha! @jrvanwhy Thanks for pointing out that the new UART HIL (as specified in the TRD) has this. Unfortunately we're still not compliant to that new specification in our actual implementation... So this is yet another good reason to push #4181.

closed 2025-01-13 by lschuermann