tock/tock #4316: Does not boot on ESP32-C3 Super Mini

BlueGradientHorizon, 2025-01-22:
Hello. I'm trying to flash and boot Tock OS on that board, but it shows no signs of life after flashing.
The board I'm tinkering with looks like this:
[image]
Windows shows device ID: `USB\VID_303A&PID_1001&REV_0101&MI_00`
I flash the ROM in WSL Ubuntu 24.04.1 and use `usbipd` tool.
Log output from `make install`:
```
esptool.py --chip esp32c3 elf2image --use_segments --output binary.hex /home/bghorizon/tock/target/riscv32imc-unknown-none-elf/debug/esp32-c3-board.elf --dont-append-digest
esptool.py v4.8.1
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.
esptool.py --chip esp32c3 write_flash --flash_mode dio --flash_size detect --flash_freq 80m  0x0 binary.hex
esptool.py v4.8.1
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (QFN32) (revision v0.4)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: d8:3b:da:34:b0:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00015fff...
Flash params set to 0x022f
Compressed 87136 bytes to 55015...
Wrote 87136 bytes (55015 compressed) at 0x00000000 in 0.8 seconds (effective 905.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

All `tockloader` commands (`listen`, `list`, `info`, etc) just time out. For example, `listen` just hangs:
```
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] No serial port with device name "tock" found.
[INFO   ] Found 1 serial port.
[INFO   ] Using "/dev/ttyACM0 - USB JTAG/serial debug unit".
[INFO   ] Listening for serial output.
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x400462e2
SPIWP:0xee
mode:DIO, clock div:1
load:0x40380000,len:0x15408
load:0x403affe0,len:0x20
SHA-256 comparison failed:
Calculated: cf21fe0e2423d4dbb27944a58b6ad0f90d97ad95d31a5d6148c13d8f3865aa10
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x40380000
<indefinite waiting>
```

I'm sorry if the board I have is actually seriously different from DevKitM-1, which is stated as supported platform and mine board is supposed to not boot. Are there at least any methods to debug kernel booting, get exception or panic messages about what's wrong in boot process?
Thanks.

bradjc, 2025-01-22:
With the preface that I have not used either board myself, I'm looking at how UART is setup on each board, and I think they are different. From [here](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-devkitm-1/user_guide.html#hardware-reference), the c3-devkit appears to have a UART-to-USB chip on board:

[image]

that I don't see on the [super mini](https://dl.artronshop.co.th/ESP32-C3%20SuperMini%20datasheet.pdf):

[image]

I'm wondering if to see the UART console output from the board you would need to connect to pins 20 and 21 on the board:

[image]

Farther down the document there is a section on serial:

[image]

It seems like you would _want_ to use serial over the existing USB header, but I'm guessing that would require implementing a USB driver in Tock. We do have other boards with other MCUs that implement serial over USB directly, but we don't (I don't think) have a USB driver for the esp32.

If you have an external UART-to-USB adapter you should be able to connect that to pins 20 and 21 on the board to see serial. Although, there might be some configuration in the kernel required to setup the UART hardware correctly, I'm not entirely sure.

bradjc, 2025-02-17:
Were you able to resolve this issue?

lschuermann, 2025-02-24:
Closing without prejudice; @BlueGradientHorizon feel free to re-open if you're still working on this.

sebinbash, 2025-07-04:
In case this is still relevant to anyone: Tock OS runs just fine when connecting via UART on pins 20, 21 and GND:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x40380000,len:0x10208
load:0x403affe0,len:0x20
load:0x403b0000,len:0x4c0
SHA-256 comparison failed:
Calculated: aa1c085a009c87b0af15d00cac02ad19cf1aae6c8100410bdcfeca85da0bc2ce
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x40380000
ESP32-C3 initialisation complete.
Entering main loop.
```