tock/tock #4326: `kernel::utilities::registers` interfaces should have different types for address and value

ZhekaS, 2025-01-27:
The current implementation, for instance:
https://github.com/tock/tock/blob/14ebb3a67d17499358afa7319fd396d735b1458a/libraries/tock-register-interface/src/registers.rs#L45-L58

implies that both the address of the register and the stored values are of type `T`, which isn't necessarily true. The address, in general should be of type `usize` (but surely can be parametrized as well), while the stored type can be any `UIntLike`. 

This is specifically important when, for example, we have byte-sized registers packed together on a 32-bit wide address space, and these registers have some side-effects on access. Or if some of them can't be accessed at all.

I believe it needs to have two separate generic parameters, one for the address type and one for the value type.

ppannuto, 2025-01-28:
Cross-ref comment for https://github.com/tock/tock/pull/4001. Off the cuff, this seems like a slightly more invasive change, and might make more sense to get rolled into the bigger rewrite [though that also comes with a longer timeframe]

jrvanwhy, 2025-01-28:
I'm definitely confused about the concern here. In the code you linked, the values are of type `T` and the address of the register is of type `&Self`. Maybe the confusion is the `self.value.get()` call? Note that [`UnsafeCell::get`](https://doc.rust-lang.org/std/cell/struct.UnsafeCell.html#method.get) returns a pointer to the value, not the value itself (this differs from `Cell`).

In general, the address should be a thin pointer (e.g. `*mut ()`) rather than `usize` wherever possible. The fact it is currently a reference type is an issue, which we are trying to address in #4001.

ZhekaS, 2025-01-28:
> I'm definitely confused about the concern here. In the code you linked, the values are of type `T` and the address of the register is of type `&Self`. Maybe the confusion is the `self.value.get()` [...]

I think you are right and I got confused. Thank you, close this issue

closed 2025-01-28 by ZhekaS