tock/tock #4333: PMP Regions can't be used to define the upper memory region

ZhekaS, 2025-01-31:
Currently, `TORRegionSpec` is defined using two `*const u8` pointers, `start` and `end`, however the `end` pointer is more like "one-past-end" because it is being written as is (with shifting by 2) to `pmpaddr` register. It means that this mechanism can't be used when the uppermost region needs to be defined. That is, when `end` is one past the 0xFFFF_FFFF address on a 32-bit system. 
Similar problem exists when we want to have a single NAPOT region to cover the whole memory space (for instance when we want a "default" PMP permission for everything). In that case the region is defined by it's start address and a size expressed as `usize`. But `usize` can't express the full size of 32-bit space.

lschuermann, 2025-01-31:
This is indeed an issue. The `TORRegionSpec` struct has been designed this way to avoid calculating the end address from the start and length at configuration time, instead only requiring a simple bit shift for both addresses.

Following this reasoning, I propose that the `TORRegionSpec` instead internally store the `pmpaddrX` values in their encoded form, taking `start: *const u8` and `size: usize` as parameters for its constructor. Those values can then be losslessly reconstructed from the encoded register values.

I can submit a PR for this later today.

lschuermann, 2025-01-31:
> In that case the region is defined by it's start address and a size expressed as `usize`. But `usize` can't express the full size of 32-bit space.

I don't quite follow this argument. Shouldn't an address of `0` and a size of `usize::MAX` cover the whole address space?

jrvanwhy, 2025-01-31:
> I don't quite follow this argument. Shouldn't an address of `0` and a size of `usize::MAX` cover the whole address space?

Off-by-one error: the address space is 1 byte larger than `usize::MAX`.

ZhekaS, 2025-01-31:
> > In that case the region is defined by it's start address and a size expressed as `usize`. But `usize` can't express the full size of 32-bit space.
> 
> I don't quite follow this argument. Shouldn't an address of `0` and a size of `usize::MAX` cover the whole address space?

If we want a NAPOT region to cover the addresses `0-0xFFFF_FFFF`, the `start` would be `0` and the size would be `0x1_0000_0000` - which does not fit in `usize`.

lschuermann, 2025-01-31:
Yeah, I don't think I was fully present when writing that comment, my bad. 🤦

I really want to avoid introducing integer types larger than usize in this interface. I would've proposed two constructors, where one is either specifically designed to cover the full address space or accepts a raw (but, in the case of TOR) validated register value(s). However, then in turn methods such as `len` may return values that we can't represent in a usize.

I'll look into removing those methods perhaps.

ZhekaS, 2025-01-31:
> Yeah, I don't think I was fully present when writing that comment, my bad. 🤦
>
> I really want to avoid introducing integer types larger than usize in this interface. I would've proposed two [...]

Perhaps we could have an enum for sizes? Because in NAPOT it can be only a powers of 2, so we have only 31 variants (could be off-by one here as well :) ). For 64 bit it would double. 
Or we can be a bit less friendly and accept the corresponding power of 2 as integer.

lschuermann, 2025-02-01:
> Perhaps we could have an enum for sizes? Because in NAPOT it can be only a powers of 2, so we have only 31 variants (could be off-by one here as well :) ). For 64 bit it would double.

That's an interesting idea, but it would also necessitate a conversion method when such regions are dynamically created, e.g., for processes through the MPU trait.

Ultimately, I strongly believe that we should expose the full capabilities of the underlying hardware, while also not making the common case more complex or inefficient.

To that end, I have PRed a solution that should achieve this compromise: by representing PMP region specifications using the underlying `pmpaddrX` values, we can cover _exactly_ the behavior that the hardware supports (in fact, I also added a check for the WARL-0 uppermost 10 bits for RV64 pmpaddrX CSRs). At the same time, we can retain the constructors for such regions from regular `address + size` and `[start; end]` address pairs. These constructors are suitable for the majority of cases and convenient in board initialization routines and for dynamic (process-related operations).

I am deliberately punting on the problem of exposing this information back in a form that is not raw `pmpaddrX` CSR values. When the time comes that we need those accessors we can re-add them, either using 64-bit values (which luckily work for 64-bit targets because of the 56-bit physical address space restriction!), or another enum-based approach.

Hope this solves your issue @ZhekaS, I'd appreciate a review of #4334.

closed 2025-04-22 by bradjc