tock/tock #4371: App crashing allows potential sharing of kernel memory

vrindisbacher, 2025-03-14:
Hi - I *believe* that there is a potential issue with the current read/write allow sharing implementation. I believe that you can achieve the same affect as: https://github.com/tock/tock/issues/1141 by deliberately crashing a process whose fault policy is restart. Here is a scenario that describes the problem:

1. There is a process (let's call it process `s`) that exposes some service to other apps. 
2. There is another (let's call it process `c`) process that uses said service. 
3. Then, process `c` uses a read write allow syscall to give process `s` access to a read/write buffer.
4. Then, process `c` crashes. *If* the fault policy for the app is to restart, it seems a problematic case arises. 

Looking at `try_restart`, it calls `reset` which [resets the app break **AND** the high water mark to their initial values](https://github.com/tock/tock/blob/1eccd0b53d86269fb396038f83a3b88ec72c920e/kernel/src/process_standard.rs#L2109). Unfortunately, at this point there will be a read/write buffer that is unguarded by the high water mark, and the grant region might grow into this buffer. This would allow process `s`, the server, to access kernel memory.  

From reading the [docs on restart policies](https://book.tockos.org/course/policies.html?highlight=restart#custom-process-fault-policy), I understand that not every app may be given a restart fault policy but I think this is still problematic and should be protected against.

bradjc, 2025-03-17:
This is an issue with the IPC implementation, correct? I agree this sounds like an issue to go with the others: #1993

vrindisbacher, 2025-03-18:
Ya I think it is an issue with the IPC implementation.