tock/tock #4383: docs.tockos.org doesn't generate entries from doc comments on macros

ppannuto, 2025-03-25:
I was going to point someone to our [quite well-explained `static_init!`](https://github.com/tock/tock/blob/master/kernel/src/utilities/static_init.rs#L5), but noticed that the [docs.tockos.org page for `static_init`](https://docs.tockos.org/kernel/utilities/static_init/index.html) only documents the function, not the macros.

Hopefully this is just a matter of finding the right flag to get rustdoc to generate entries for macros as well as functions.

bradjc, 2025-03-26:
There is https://docs.tockos.org/kernel/macro.static_init

ratmice, 2025-04-01:
Given the objections raised on my PR I don't really see this one being actionable without changes to upstream tooling.

There are crates like:

* https://crates.io/crates/macro-vis
* https://crates.io/crates/macro-v
* https://crates.io/crates/macro_pub

But these all seem to be variations on the same theme used in the patch but involving macros.
Rather than what I tried, exporting the docs from *both* the root and the module.

These crates all seem to hide the docs in the root, and hiding the macro in the root by giving it a random unguessable name.
I don't feel that using any of those crates would overcome either of the objections raised.

ppannuto, 2025-04-01:
We have successfully proposed changes to rustdoc before, sounds like this is a case where we might need to do that again. 

We should make sure there isn't any open issue on rustdoc we can add a voice to before making a new one. Once there is something open upstream, we can tag this `#blocked_upstream` and add a tracking link.

ratmice, 2025-04-01:
Ahh, apparently this is already a feature of [macros 2.0](https://github.com/rust-lang/rfcs/blob/master/text/1584-macros.md) that works with nightly https://github.com/rust-lang/rust/issues/39412

ppannuto, 2025-04-04:
Makes sense. I'll mark this as blocked upstream until the revised macro scoping lands in stable [acknowledging here that that may well be a while].