tock/tock #4400: BLE Functionality Enhancement

mohakvni, 2025-04-08:
Our current BLE advertising driver supports basic periodic advertising and passive scanning with a pseudo-random address generation mechanism. While this setup works for basic functionality, we should add more enhancements such as support for more hardware, BLE 5.0 features, and security improvements. Here are some ideas I was thinking of:

- Extended Advertising: Enable support for larger payloads and improve buffer management for advertising data.
- Period Advertising: Allow the driver to send packets at regular intervals with data that can change between intervals.
- Improved Address Generation: Replace the fixed/pseudo-random address with a more secure, truly random mechanism.
- Active Scanning: Implement continuous scanning and active response to received advertisements.
- Duplicate Advertisement Filtering: Add filtering to avoid processing duplicate advertisements.
- Multi-board Support: Expand support beyond the NRF and Apollo3 SoCs to include other boards.

Would love to hear some feedback as well as any other ideas

bradjc, 2025-04-09:
Absolutely!

The challenge is _how_. Do we try to implement this directly in Tock? Or try to integrate an existing BLE library?

Implementing in Tock usually means defining interfaces for the functionality, so that it can be portable to multiple boards, which can difficult. Using a library means finding a suitable one, which hasn't worked out.

But, if you enthusiastic, we would be thrilled to have better BLE support in Tock!

And just in case it is helpful, here are some related issues:
- https://github.com/tock/tock/issues/2970
- https://github.com/tock/tock/issues/981

ppannuto, 2025-04-09:
All of these would be valuable for Tock. There's varying complexity to implementing them in practice (loosely, your list is probably already ordered in my sense of "expected difficult to implement").

If the question here is really "where should I start?", I would suggest either Extended Advertising or Periodic Advertising, as they are likely an order of magnitude easier than anything else on the list.

alevy, 2025-04-09:
> Active Scanning: Implement continuous scanning and active response to received advertisements.

Perhaps responding to active scans instead?

This seems by far the highest product of tractable + valuable to me. Active scanning is part of the first step towards connection-oriented communications anyway and requires solving _just enough_ of the timing issue to be both tractable and show promise.

Doing this on the NRF seems like a good goal (rather than multiple chips).

alevy, 2025-04-09:
If you look at issue [#981](https://github.com/tock/tock/issues/981) that @bradjc posted, you'll see a link to a PR that has completely functional connections, but was just on too old a version of Tock to actually merge reasonably.

mohakvni, 2025-04-15:
Thank you @bradjc @ppannuto @alevy. I agree that implementing this directly in tock will be a challenge, but I am very enthusiastic about contributing and learning about tock, so I am up for the challenge. I think as a first steps, I can begin with @alevy suggestions to implement active scan responses on the nrf and start looking at [#981](https://github.com/tock/tock/issues/981).