tock/tock #4417: Enhance RiscV PMP to support minimum granularity greater than 4 bytes

ZhekaS, 2025-04-29:
The PMP on Risc-V can have a minimal granularity other than currently supported 4 bytes. For example, SiFive cores that have MMU (not necessarily used, just present) are forced to have the minimal granularity of a page size (that is 4096 bytes).
I am not sure of the whole scope of the changes required, but I imagine that simply rejecting region specifications that do not fit in the granularity restriction at a construction time should suffice.
The minimum granularity can be either hardcoded (a generic parameter to XXXPmpRegionSpec?? A constructor parameter?) or detected as documented in "The RISC-V Instruction Set, Manual: Volume II, Privileged Architecture".

Other than that I believe the extra care should only be needed for boards that actually have the restriction. That is designing memory layouts that are aligned to the granularity sizes. Not sure what are the implications for memory management of processes.

lschuermann, 2025-04-30:
Just a stub comment for now, I want to get back to this quickly but will be busy through Friday (2025-05-02). Please feel free to ping if I don't get back to this! :)

ZhekaS, 2025-05-06:
I think it can be either implemented in a more elegant way, but it will be more "intrusive" in a sense that will require changes across the different board files, or it can be implemented less nicely with leaving the board files as is. What is the preferred approach in general?

ZhekaS, 2025-06-23:
@lschuermann 
> Just a stub comment for now, I want to get back to this quickly but will be busy through Friday (2025-05-02). Please feel free to ping if I don't get back to this! :)

Pinging as requested ;)

closed 2026-09-27 by jrvanwhy