tock/tock #4418: RFC: Policy on creation of capabilities in components

ppannuto, 2025-04-30:
One of the purposes of the capability mechanism is to highlight to board/platform authors where "sensitive" things occur and to restrict access to them.

#4411 aimed to use `ProcessManagementCapability` in this way to highlight that including the `ProcessInfo` capsule on a board would, without filtering or some other protections, by default allow _any_ userspace process to control any other process.

By creating the capability in the component, however, this ends up looking like "any other component" in a board. @alevy suggested that it probably makes sense to disallow capability creation in components generally, and to always require boards to pass references to components. That's a more invasive change, however, and merits a bit more discussion — hence this issue.

---

To help seed discussion, I'll add this:

**Concrete Initial Proposal / Default Action:** We should restrict the creation of capabilities _exclusively_ to top-level board files*. We should document this policy and explain its rationale. We should refactor existing components to remove all capability creation, and lift them to board files.

*Open Question: Can we do this mechanistically (through visibility of creation) somehow?

brghena, 2025-04-30:
I agree that components should not be able to automatically create capabilities. I do not believe creation capability MUST be limited to top-level board files. There could be other acceptable mechanisms.

bradjc, 2025-04-30:
I don't think we have any capsules that take a grant as an argument to the component `new()` function. Someone should try that to see what it looks like to help inform this discussion. I wonder if it isn't harder than it would seem.

ppannuto, 2025-04-30:
> I do not believe creation capability MUST be limited to top-level board files. There could be other acceptable mechanisms.

Do you have any specific use cases in mind? From a quick grep, currently only `boards/` ever create capabilities. I can't immediately think of an example where the ergonomics of creating it outside the top-level board outweighs the transparency and intent it forces on board authors.

That, of course, is quite likely a statement more of the limits of my own creative thinking, but I wonder if we shouldn't start with a policy of only top-level boards until a compelling use case arises that requires us to relax the policy (akin to how we eventually relaxed our third-party dependencies for crypto stuff).

brghena, 2025-04-30:
> Do you have any specific use cases in mind? 

I could imagine a helper (maybe a component) that creates capabilities, and then the board file manually distributes them. For example, as arguments to components. So it would still be on the board file to direct their use, but not necessarily explicitly to create them.

brghena, 2025-04-30:
> I don't think we have any capsules that take a grant as an argument to the component new() function. 

You mean "take a **capability**" right?

bradjc, 2025-04-30:
> > I don't think we have any capsules that take a grant as an argument to the component new() function.
> 
> You mean "take a **capability**" right?

No, I mean grant. 

I guess there are two options, capabilities can be used in components but must be passed in, or the operation that requires a capability must be done in main. I suppose if we are only considering the former then there is no need to explore how grants can be passed as arguments.

ppannuto, 2025-04-30:
> capabilities can be used in components but must be passed in

I believe that was the design point we came to on the call this morning.

bradjc, 2025-04-30:
Does Rust care if you create a struct and implement all capabilities for it, and then use it for something that only needs one?

Related, do we permit one object with multiple capabilities or a separate argument for each capability? I assume for clarity we would want each required capability enumerated separately.

ppannuto, 2025-04-30:
> Does Rust care if you create a struct and implement all capabilities for it, and then use it for something that only needs one?

I believe that would be accepted. We implement capabilities as traits, and interfaces require that you "pass an object which implements the necessary trait" — that says nothing about what other traits you might also happen to implement.

(checking myself...)

```diff
diff --git a/boards/hail/src/main.rs b/boards/hail/src/main.rs
index d35a40714..008840e79 100644
--- a/boards/hail/src/main.rs
+++ b/boards/hail/src/main.rs
@@ -33,6 +33,10 @@ pub mod io;
 #[allow(dead_code)]
 mod test_take_map_cell;

+struct MegaCap;
+unsafe impl capabilities::ProcessManagementCapability for MegaCap {}
+unsafe impl capabilities::MemoryAllocationCapability for MegaCap {}
+
 // State for loading and holding applications.

 // Number of concurrent processes this platform supports.
@@ -509,7 +513,7 @@ unsafe fn start() -> (
         ipc: kernel::ipc::IPC::new(
             board_kernel,
             kernel::ipc::DRIVER_NUM,
-            &memory_allocation_capability,
+            &MegaCap,
         ),
         crc,
         dac,
@@ -552,7 +556,7 @@ unsafe fn start() -> (
         ),
         &mut *addr_of_mut!(PROCESSES),
         fault_policy,
-        &process_management_capability,
+        &MegaCap,
     )
     .unwrap_or_else(|err| {
```

(that works fine)

---

I'm not sure policy-wise. I don't know if we have any interface currently that requires more than one capability, but there is possibly something elegant is taking a single parameter for "all the capabilities I need" rather than two parameters (but maybe the benefit of being explicit and easily searchable outweighs that). Having written that run-on sentence, I would lean to a "one Cap per struct" policy I think.

lschuermann, 2025-05-06:
Forwarding some (updated / corrected) messages of mine from a Slack channel:

> I think we might want to revisit https://github.com/tock/tock/pull/3409 even before Tock 3.0. As [this example](https://github.com/tock/tock/pull/4411#discussion_r2074620163) shows, the way we currently use capabilities seems dangerous / unsound (not in the Rust language sense, but in the guarantees we want to get from them).
> 
> The problem here is that we define capabilities as `pub struct` types, on which we then implement the capability traits as an unsafe operation. The operation we want to be unsafe is the creation of a new capability (instance). However, currently, the operation that is unsound is declaring that a given type and all of its instances represents a capability.
> 
> Unfortunately, the way we define these types right now means that everyone who can name them can get a fresh capability safely, without an additional use of unsafe. For a conceptually similar type, the Rust compiler source uses a private field to work around this (with a `pub(crate)` annotation), but even that won't work for our use-case. Because we place different (mutually distrustful) components in a single Rust crate (like `capusules-extra` or `components`), then all other capsules / modules in that same crate will be able to get new copies of a capability without additional unsafe.
> 
> @jwnrt's solution is much better, as it moves the `unsafe` to the constructor of this type, and defines the `Capability` type itself with a private field in the kernel module -- making sure that other crates cannot simply invent new copies of the type without going through the unsafe constructor.

hudson-ayers, 2025-05-14:
> One thing worth noting, that might help "implement" this policy, is that currently `create_capability!` does _not_ actually require an `unsafe` context---so, most concerning, it can be called from a capsule to create their own capabilities.

This is not true, `#![forbid(unsafe_code)] still catches uses of unsafe from called macros:
```
hayers@svdi-u8q-321:~/code_repos/tock/boards/imix $ make
   Compiling capsules-core v0.2.3-dev (/home/hayers/code_repos/tock/capsules/core)
error[E0453]: allow(unsafe_code) incompatible with previous forbid
  --> capsules/core/src/alarm.rs:54:36
   |
54 |         let main_loop_capability = create_capability!(capabilities::MainLoopCapability);
   |                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ overruled by previous forbid
   |
  ::: capsules/core/src/lib.rs:5:11
   |
5  | #![forbid(unsafe_code)]
   |           ----------- `forbid` level set here
   |
   = note: this error originates in the macro `create_capability` (in Nightly builds, run with -Z macro-backtrace for more info)

```

alevy, 2025-05-14:
@hudson-ayers so this is relying not on the the use of `unsafe` in the macro, but rather the explicit presence of an `allow(unsafe_code)` in the macro implementation. That allow isn't necessary for anything, and if you remove it, your code above will compile (and nothing else will break).

However, I think this is fine. My remedy is simply to document in the doc-comment for the macro that this is the safety invariant---it _has_ to have this `allow` directive in the implementation.

bradjc, 2025-05-15:
> [@hudson-ayers](https://github.com/hudson-ayers) so this is relying not on the the use of `unsafe` in the macro, but rather the explicit presence of an `allow(unsafe_code)` in the macro [...]

This is completely bonkers and is completely changing my understanding of what `#![forbid(unsafe_code)]` is doing for us.

How can we (safely) include macros in Tock at all?

ppannuto, 2025-05-15:
🤯 ... _wow_ does Rust make it difficult to contain `unsafe` in practice...

Does this only "work" because the macro happens to be defined in crate that does allow the use of unsafe? i.e., if the `create_capability!` macro were defined in the capsules crate, could you still "sneak in" `unsafe`?

ppannuto, 2025-05-16:
It appears the answer to that is yes, and it's a deliberate design choice: https://github.com/rust-lang/rust/issues/110613#issuecomment-1516813435

bradjc, 2025-05-20:
We've run into this issue:

```rust
pub struct PMCapability;
unsafe impl capabilities::ProcessManagementCapability for PMCapability {}
unsafe impl capabilities::ProcessStartCapability for PMCapability {}

type ProcessInfoDriver = capsules_extra::process_info_driver::ProcessInfo<PMCapability>;

let process_info = components::process_info_driver::ProcessInfoComponent::new(
    board_kernel,
    capsules_extra::process_info_driver::DRIVER_NUM,
    PMCapability,
)
.finalize(components::process_info_component_static!(PMCapability));
```

Where we need to know the type for the macro and we need to pass the object to the component. Not sure how this will work with macros.