tock/tock #4469: FieldValue should support BitOr and BitOrAssign

BartMassey, 2025-06-17:
Right now, `fields::FieldValue` uses `Add` and `AddAssign` to combine field values. It seems more intuitive to use `BitOr` and `BitOrAssign` for this, since internally it is using bitwise ors. Adding these implementations should not break backward compatibility.

Let me know if you want a PR. Thanks much for this crate.

ppannuto, 2025-06-17:
Ah, I was curious why we didn't support that already. Answer is that it looks like it's a relatively new trait:

[image]

I think it's something we would happily add support for, and a PR would certainly be appreciated!

lschuermann, 2025-06-17:
> Answer is that it looks like it's a relatively new trait:

[Rust 1.8.0 was released in 2016](https://releases.rs/docs/1.8.0/), so not particularly new. Maybe you read 1.80.0 instead? 😄 

From memory and without having looked at the code, I thought that we're using add because we're "combining" different fields. While that's implemented with a bitwise-or underneath, it's a logical concatenation of different fields into a single register value (which is more analogous to an "add" operation).

alexandruradovici, 2025-06-17:
I kind of agree with @lschuermann, the add seems more intitive than bitwise or, as we practically add fields to the register.

Moreover, we actuall set the value of the field, which is a kind of a "zeroing and bitwise or". Another example is using F::CLEAR that performs a bitwise and.

bradjc, 2025-06-19:
This is one where it has always been that way, where my recollection is the original developer found it more intuitive to use `+`. Having both would make code readers have to interpret if there is a difference, and changing all `+` to `|` seems like significant effort. Probably best to leave it.

lschuermann, 2025-06-19:
I agree with @bradjc. While I'm sympathetic to the concerns raised by @BartMassey (and appreciate the suggestion!), I do think that sticking to the current operator is both better than introducing a new, aliased one (for the reasons @bradjc mentioned), or introducing a breaking change and switching to bitwise-or.

Closing this without prejudice, if others chime in we can gladly reopen.