tock/tock #4485: Please publish a new version of `tock-registers` to allow const expressions in offsets.

aarkegz, 2025-06-27:
`tock-registers` is an amazing crate that I use it every time I write a driver in Rust. But today I found that using const expressions as offsets or size in `register_structs!` results in compiling errors originating from `test_fields!` which tried to `concat!` the const expression with other strings. It's kind of frustrating because using const expressions as offsets or sizes of `register_structs!` is more readable and maintainable.

Luckily, the commit https://github.com/tock/tock/commit/ea37a2e12f12687b25d7f886b1f12572e0331bcb has already fixed this bug, but the latest release of `tock-registers` now (0.9.0) does not include this fix. Could you please publish a newer version of `tock-registers`? My sincerest thanks.

lschuermann, 2025-06-27:
We've been hesitant to do a new release of `tock-registers` as we've been working on a new version that has some major API breakage, and wanted to release the last `v0.x.0` just before we introduce that breakage.

However ... that's been a long while coming at this point, so I'm totally sympathetic to just tagging `v0.9.1` today (assuming that we don't have any breakage between the current HEAD and `v0.9.0`).

@ppannuto Thoughts?

brghena, 2025-06-27:
I see no harm in additional releases, especially if a downstream user has a reason for it.

lschuermann, 2025-06-27:
Opened #4486 for this.

lschuermann, 2025-06-28:
@aarkegz v0.10.0 is published to crates.io: https://crates.io/crates/tock-registers/0.10.0

Please let us know if there are any other issues with `tock-registers` or this release. 😃