tock/tock #4550: Question about timeslice value set in MLFQ scheduler

tonywangxy, 2025-08-06:
Hi Tock team, 

I have a question about below code snapshot in mlfq.rs : 

`        node_ref
            .state
            .us_used_this_queue
            .set(self.last_timeslice.get() - execution_time_us);
`

`execution_time_us` here means the time that this process already consumes, and `last_timeslice` means the total timeslice allocated for this process, right? 

If so, `(self.last_timeslice.get() - execution_time_us)` means the left timeslot for this process, which is not aligned with the definition of `us_used_this_queue`. 

Please help to confirm. Thanks a lot!

Tony

brghena, 2025-08-06:
That does look like a bug to me.

After staring at this for a while, I think it's not just a bug on that line but a bigger issue. MLFQ should track total runtime for each job. When scheduled, the job's timeslice should be set to `self.get_timeslice_us(queue_idx)` minus that total runtime. Then if the timeslice ever expires, the job is demoted a level and total runtime should reset to zero.

If you're interested, we would be willing to accept a PR for this. I suspect no one has been using the MLFQ scheduler.

tonywangxy, 2025-08-11:
Thanks @brghena for your confirmation!
Actually, I am a quite new learner for Rust/Tock. I don't think I have the ability to raise a PR to fix this issue now. Can any expert help to fix this issue?