tock/tock #4584: Release 2.3

bradjc, 2025-09-06:
Goals for 2.3:
- rv32 qemu support for tutorials
- remove `static mut` (use `SingleThreadValue`)

Done for 2.3:
- x86

lschuermann, 2026-09-10:
As per our recent core call discussion and #5169, I've branched off `release/2.3`: https://github.com/tock/tock/tree/release/2.3

We can start testing on this branch now. I will shortly add a comment with an updated list of tests for the nRF52840DK board which we can use as a template (although other tests may need to be added for other boards).

bradjc, 2026-09-10:
I've tested:

- `esp32-c3-devkitM-1`: c_hello, blink, RNG, sensors apps work. I fixed a bug in the memory layout which prevented running 3+ apps concurrently.
- `wm1110dev`: It can connect to the Things Network via LoRa and sample temperature and humidity sensors.

bradjc, 2026-09-10:
I've also tried a basic version of the dynamic apps and policies tutorial on the nrf52840dk and it works as expected.

bradjc, 2026-09-11:
Can we put a deadline of this Wednesday 9/16 on this release?

bradjc, 2026-09-15:
I have run a bunch of tests on microbit, including the sensors, buzzer, microphone, rng, and BLE. All of the issues I've found have open PRs. I'm happy with that level of coverage on the microbit.

brghena, 2026-09-15:
I ran the RP2040. With https://github.com/tock/tock/pull/5156 and https://github.com/tock/tock/pull/5175 I can interact with it from MacOS and successfully run the `c_hello` app.

Without those, I can't program apps onto the RP2040, and it doesn't enumerate in USB so I can't communicate with it. So, we have to decide whether to include those in a Tock 2.3 release or not.

alexandruradovici, 2026-09-16:
I suggest we do as the pico board is easy to get and use.

lschuermann, 2026-09-23:
I ran into an issue while trying to test https://github.com/tock/tock/pull/5194 and reproduce the original issue on real hardware (not just from a hypothetical analysis of the assembly).

On that board, and many others, any kernel hard-fault will cause an infinite loop of hard faults. This is because the panic writer uses the legacy path and creates the regular Uart driver, which internally uses SingleThreadValue, which in turn correctly returns `None` in a hard fault context.

This means that, from a cursory reads, currently all of these boards get stuck in an infinite loop without printing anything on a hard-fault:

- `nucleo_f429zi`
- `nucleo_f446re`
- `stm32f412gdiscovery`
- `stm32f429idiscovery`
- `weact_f401ccu6`
- `stm32f3discovery`
- `lora_e5_mini`
- `hail`
- `imix`
- `hifive1`
- `hifive_inventor`
- `redboard_redv`
- `arty_e21`
- `litex/arty`
- `litex/sim`
- `opentitan/earlgrey-cw310`

I don't think there's a cheap way for us to work around it, other than carry on the valuable work that @bradjc has been doing to port more boards over to the new panic printing approach. I vote we mark this as a known issue for the upcoming 2.3 release.

lschuermann, 2026-10-01:
Given the substantial amount of bugs fixed since the initial branch-off point, and the evolution of the `master` branch in that time, we've now fast-forwarded the `release/2.3` branch to current `master`, excluding only the merge commit of #5184. Apart from the version, these are now identical.

Following our release document, I will run another set of tests on this new branch. Notwithstanding any unknown and/or critical bugs, I will bump the kernel version on that branch to the full release, write release notes, and then tag the release in the next few days.

bradjc, 2026-10-08:
https://github.com/tock/tock/releases/tag/release-2.3.0

closed 2026-10-08 by bradjc