tock/tock #4591: x86 EOI timing issues

reynoldsbd, 2025-09-10:
I'm working on adding support for using Virtio devices on the x86 architecture using PCI as the transport backend. Along the way I've run into some issues with the way EOI signals are dispatched by the current implementation. Tagging @alexandruradovici and @ioanaculic, and I'd love to get input from anyone else who uses the x86 port.

---

The first issue is relatively straightforward: The existing EOI code does not properly handle when an interrupt is cascaded from PIC2 to PIC1. In such cases, the EOI should be sent to _both_ PIC instances, but today we are only sending it to PIC2.

https://github.com/tock/tock/blob/5f606cc9352797b50262b4970ba169d8336aa6c4/chips/x86_q35/src/pic.rs#L104-L108

Fix here is a simple one-liner to send EOI to both PICs, and I'm preparing a PR which I will submit soon.

---

The second issue is a bit more subtle and relates to edge- versus level-triggered interrupts. The global handler for external interrupts is currently hard-coded to unconditionally issue an EOI to the PIC after updating the `InterruptPoller`:

https://github.com/tock/tock/blob/5f606cc9352797b50262b4970ba169d8336aa6c4/chips/x86_q35/src/interrupts.rs#L20-L25

This works great for edge-triggered interrupts. The interrupt itself is suppressed, and the kernel loop can come along and service the interrupt synchronously. Today, all the legacy PC/BIOS devices in the x86_q35 crate (PIT, serial port) are edge-triggered.

But for level-triggered interrupts, such as those produced by PCI devices, we cannot issue the EOI directly from the ISR because at that point the device driver has not yet had an opportunity to service the hardware. The interrupt line remains asserted, and the ISR keeps firing and floods the system. 💀 

I can think of two obvious solutions, and neither is perfect:

1. Move EOI from the ISR into the chip's `service_pending_interrupts()` routine. This works for me locally, but it effectively blocks the kernel loop from servicing more than a single interrupt at a time.
2. Within the ISR, fire off the EOI _conditionally_ based on some heuristic or possibly some chip- or board-defined configuration. Maybe a global bitfield indicating which interrupts are edge or level? Feels really clunky 😕

HMiyaziwala, 2025-09-10:
I think we should go with the latter proposal - some conditional logic based on the nature of the interrupt is inevitable given that level and edge are handled differently.

Here is a design proposal...

We can introduce a trigger enum:

```rust
 pub enum Trigger {
    Edge,
    Level,
}
```
And create a table (or bitmask) and pass this to the PIC's `init` function from the board.
We also introduce an API to mask and unmask a given interrupt. Then we can:

```rust
unsafe extern "cdecl" fn handle_external_interrupt(num: u32) {
    match IRQ_DESC[num] {
        Trigger::Edge => {
            InterruptPoller::set_pending(num);
            pic::eoi(num);
        }
        Trigger::Level => {
            // mask out to prevent storm
            pic::mask(num);
            InterruptPoller::set_pending(num);
            // no EOI here
        }
    }
}
```

and the handler then calls complete (conditionally?):
```rust
pic::eoi(num);
pic::unmask(num);
```

alevy, 2025-09-10:
> Another approach could be to adopt a model where peripheral drivers are ultimately responsible for sending EOI

This would be most inline with how ARM and RISC-V (implicity) do this. On those architectures, typically, the Interrupt Status Register in the NVIC (or equivalent) is de-asserted by the peripheral whenever the condition is handled---e.g., the UART de-asserts the "input available" interrupt when the driver reads all the values from the UART's input buffer register.

In the top-half interrupt handlers, on ARM for example, we disable the interrupt and rely on the driver to re-enable the interrupt when it's done.

This puts a _bit_ more burden on the driver, but also the driver is the thing that actually _knows_, in a peripheral-specific manner, when it's done and when interrupts would be meaningful again.

I'm not sure if/how to do the second part---disabling interrupts until the driver says so---on x86

reynoldsbd, 2025-09-16:
These issues have been addressed by #4596

closed 2025-09-16 by reynoldsbd