tock/tock #4600: Device Tree (dts/dtsi) Support for Board Definition and Hardware Abstraction

yoavfilin, 2025-09-16:
Tock OS currently requires all board and hardware configuration to be written in Rust source files. This limits portability, reusability, and maintainability, especially when compared to other embedded OS like Zephyr and Linux, which use Device Tree files for hardware description and abstraction.
Support for Device Tree would enable easier board bring-up, hardware portability, and potentially allow leveraging the large ecosystem of existing Device Tree files from Linux and Zephyr.

I would like to request support in Tock OS for hardware description via Device Tree files, similar to Zephyr OS. This would include:
- The ability to describe memory resources, peripherals, pins, and board configuration in dts/dtsi files instead of source code.
- A build step or tooling to automatically generate Rust code or configuration from Device Tree sources.
- Documentation and examples for using Device Tree with Tock OS boards.

alexandruradovici, 2025-09-20:
There was an attempt to do something similar, please take a looks at https://github.com/tock/tock/pull/4153. It is not exactly device tree, but close enough.

yoavfilin, 2025-09-21:
Why was is abandoned?

bradjc, 2025-09-30:
I agree that you can see why something like a device tree is attractive, and I'm inclined to believe that having something like a device tree would be an improvement for Tock. However, I think the most salient reason we don't have something like device tree is we don't know how to build it.

Our `Component` pattern is the closes thing we have. While it is still Rust code, it does share some similarity in that (well-written) components abstract resource configuration to just specifying what hardware should be used for what functionality.

Some other considerations that have led to our current main.rs files:
1. main.rs includes more than just hardware mapping. It is also how we configure the kernel and what kernel components we include. I would guess that by lines of code, more code is spent configuring the layering of various software stacks than defining hardware. This ends up being somewhat complex to get all of the Rust types to be correct.
2. I'm always a bit wary of adding DSLs. Something widely used like device tree is probably fine as we wouldn't be asking tock developers or users to learn a bespoke language. But, just sticking to Rust makes tock more accessible to more people.

-------

Just speaking for myself, I think the combination of:
1. Components
2. Declared `type` aliases in components
3. A liberal use of `type` aliases in main.rs

strikes a good balance of somewhat readable main.rs files, reusable main.rs files that can be copied to create a new board, flexibility to fully customize a kernel, and reasonable maintenance burdens for the Tock project.

For example, instantiating a full key-value stack looks like this:

https://github.com/tock/tock/blob/f62f0854b37ec4a49c6789d4e36febda7299c424/boards/nordic/nrf52840dk/src/lib.rs#L736-L795

But, I'm not saying it's the _best_ way. There very well may be something that is better.

bradjc, 2025-09-30:
> Why was it abandoned?

The tool required a fair bit of customization to support each chip and capsule in Tock, and it wasn't immediately obvious how to make it more scalable to maintain over time.

bradjc, 2025-10-02:
Right, something like https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/arm/nordic/nrf52840.dtsi describes the hardware, but doesn't include something like userspace should have access to the Screen driver which uses the ScreenSplit virtualizer which provides a 128x32 pixel view to userspace and uses the ScreenX driver which uses I2C, for example.